khoih-prog/AsyncHTTPSRequest_Generic #3: empty response

DavidAntonin, 2022-01-10:
I filled up my wifi ssid/pass in example but I did not get response, only:
**************************************

**************************************

khoih-prog, 2022-01-10:
Sorry, everything is OK here.

```
Starting AsyncHTTPSRequest_ESP using ESP32_DEV
AsyncTCP_SSL v1.1.0
AsyncHTTPSRequest_Generic v1.2.0
Connecting to WiFi SSID: HueNet1
........
AsyncHTTPSRequest @ IP : 192.168.2.105

**************************************
abbreviation: EST
client_ip: 216.154.35.105
datetime: 2022-01-10T14:07:53.434551-05:00
day_of_week: 1
day_of_year: 10
dst: false
dst_from: 
dst_offset: 0
dst_until: 
raw_offset: -18000
timezone: America/Toronto
unixtime: 1641841673
utc_datetime: 2022-01-10T19:07:53.434551+00:00
utc_offset: -05:00
week_number: 2
**************************************
```

You have to specify a lot more information, such as 

> Please ensure to specify the following:
> 
> * Arduino IDE version (e.g. 1.8.19) or Platform.io version
> * `ESP8266`,`ESP32` or `STM32` Core Version (e.g. ESP8266 core v3.0.2, ESP32 v2.0.2 or STM32 v2.2.0)
> * Contextual information (e.g. what you were trying to achieve)
> * Simplest possible steps to reproduce
> * Anything that might be relevant in your opinion, such as:
>   * Operating system (Windows, Ubuntu, etc.) and the output of `uname -a`
>   * Network configuration

Please remember next time to follow the instructions in [**Issue: Bug report**](https://github.com/khoih-prog/AsyncHTTPSRequest_Generic/issues/new?assignees=&labels=&template=bug_report.md&title=), otherwise **your post will be deleted** right away.

closed 2022-01-10 by khoih-prog

DavidAntonin, 2022-01-11:
> Komu: khoih-prog/AsyncHTTPSRequest_Generic <AsyncHTTPSRequest_Generic@
> noreply.github.com>
> 3)
> "

I tried it several times

Starting AsyncHTTPSRequest_ESP using ESP32_DEV

AsyncTCP_SSL v1.1.0

AsyncHTTPSRequest_Generic v1.2.0

Connecting to WiFi SSID: SOSNbk

............

AsyncHTTPSRequest @ IP : 192.168.5.49

**************************************

**************************************

HHHHHH

**************************************

**************************************

ulr: "https://worldtimeapi.org/api/timezone/America/Toronto.txt" is 
accesable from web browser without problem - notebook on the same WiFi

Windows 10 64b 21H1

Arduino IDE 1.8.19

ESP32 2.02

Board: ESP32 DEVKIT V1 DOIT

Stock example code - only changed SSID and password

VDSL2 connection 100/25

---------- Původní e-mail ----------

<code>Starting AsyncHTTPSRequest_ESP using ESP32_DEV

</code>

   
      
   
(https://github.com/khoih-prog/AsyncHTTPSRequest_Generic/issues/new?assignees=&labels=&template=bug_report.md&title=)

(https://github.com/khoih-prog/AsyncHTTPSRequest_Generic/issues/3#issuecomment-1009259031)
or Android
"

khoih-prog, 2022-01-11:
I'm sorry I couldn't duplicate and everything is OK here. You have to check your network / router settings and turn on better debug option to understand what's really happening there. 

```
#define _ASYNC_TCP_SSL_LOGLEVEL_    4
#define _ASYNC_HTTPS_LOGLEVEL_      4
```

I'll investigate only after you'd prove this is a bug of the library.

DavidAntonin, 2022-01-26:
I now have a functional sending of requests. I found that in order to send another request, I have to wait for the previous one to reply. I have solved this with the request queue and it works reliably. However, the problem occurred when I wanted to send a POST request after a GET request and vice versa. I can send multiple requests of the same type until I send a request of a different type - I get a "Can't send bad request" error (request.open = false)

khoih-prog, 2022-02-03:
Hi @DavidAntonin 

I'm currently have no time to spend on this issue yet.

Can you try by

1. Using ESP32 core v2.0.1 (not v2.0.2). Some HTTPS was reportedly causing issue in [**Connection problem via https ESP32S2** #6117](https://github.com/espressif/arduino-esp32/issues/6117#issuecomment-1020594690)

then post your test result here.

khoih-prog, 2022-02-03:
More to try and digest

1. [**WiFiClientSecure handshake fails after client.stop(). Regression in 2.0.2** #60770](https://github.com/espressif/arduino-esp32/issues/6077)

> As a workaround for the issue, one can add client.setHandshakeTimeout(30); before every connection attempt.

2. [**ESP32 core 2.0.2 does not work (2.0.1 does)** #270](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/issues/270)
 

> A temporary fix that worked for me was to change line 266 in the ssl_client.cpp (in the WiFiClientSecure library) file from:
> 
> ```
> if((millis()-handshake_start_time)>ssl_client->handshake_timeout)
> ```
> to:
> 
> ```
> if((millis()-handshake_start_time)>(ssl_client->handshake_timeout != 0?ssl_client->handshake_timeout:120000))
> ```
> (So I manually set a timeout if the handshake_timeout is zero.)

khoih-prog, 2022-02-25:
> I found that in order to send another request, I have to wait for the previous one to reply. 

Check your related issue [**Pushover https post request** #21](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/issues/21#)

khoih-prog, 2022-02-26:
> I now have a functional sending of requests. I found that in order to send another request, I have to wait for the previous one to reply. I have solved this with the request queue and it works [...]

Adapt the new example [**AsyncHTTPSRequest_ESP_Multi**](https://github.com/khoih-prog/AsyncHTTPSRequest_Generic/tree/main/examples/AsyncHTTPSRequest_ESP_Multi) of the new [AsyncHTTPSRequest_Generic releases v1.4.1](https://github.com/khoih-prog/AsyncHTTPSRequest_Generic/releases/tag/v1.4.1)

In the current design, each **AsyncHTTPSRequest** instance can only be use for **one address and one type of request** (GET, POST, etc,).

**It's more complex to completely rewrite the library, than to write the core using multiple instances.**

So I don't think I'll spend the time to fix the library in the near future.

Good Luck,

DavidAntonin, 2022-02-26:
Hi @khoih-prog 
Thank you four your time and example. I will adot it to my program
Have a nice day