khoih-prog/AsyncHTTPSRequest_Generic #10: Delay on Done

glennswest, 2022-03-28:
### Describe the bug

Doing automated update of web content on my board, and its working, but I see a consistent delay on the done state for the download.

### Steps to Reproduce

Download a "list" of files, chaining next file off the complete of last. using datacb to handle any amount of data, see multi-second delay on the done after the last datacb. (About 6 seconds)

14:43:04.400 -> Starting download for: index.html
14:43:04.400 -> 
14:43:05.031 -> Removing index.html
14:43:05.677 -> Remove done
14:43:06.321 -> Opening file
14:43:06.976 -> File Open
14:43:07.572 -> Send Download Request
14:43:08.214 -> sendDownloadReq
14:43:08.214 -> Sending request
14:43:08.214 -> https://raw.githubusercontent.com/glennswest/i2cgateha/main/contents/index.html
14:43:09.408 -> Download done callback index.html - state 1
14:43:09.620 -> Download done callback index.html - state 2
14:43:09.620 -> Download done callback index.html - state 3
14:43:15.393 -> Download done callback index.html - state 4
14:43:15.393 -> Download complete: index.html - Remainer: 0
14:43:15.393 -> No Downloads

Source: https://github.com/glennswest/i2cgateha/blob/main/I2CGateHa.ino

### Expected behavior
Done immediate after data complete

### Actual behavior
Multi-second delay

### Debug and AT-command log (if applicable)

https://gist.github.com/glennswest/2911ae27fa427a9176382c5cf390b01b

Please ensure to specify the following:
M5Stack ESP32
Arduino 1.8.16

khoih-prog, 2022-03-29:
Hi @glennswest 

Please post the [**Minimal, Reproducible Example**](https://stackoverflow.com/help/minimal-reproducible-example) and if possible locate the bug and/or propose the fix as you did in previous issue.

I'm sorry I don't have time to recreate your environment / settings to duplicate this issue.

Also be aware that TLS/SSL requires quite some time to start a new client / job due to heavy encryption/decryption, depending on the board.

Check [**Resources for TLS/SSL**](https://github.com/OPEnSLab-OSU/SSLClient#resources)

```
Note: client.connect("www.arduino.cc", 443) can take 5-15 seconds to finish on some low-power devices. 
This an unavoidable consequence of the SSL protocol
```

and

```
In addition to the above, most embedded processors lack the sophisticated math hardware commonly found in 
a modern CPU, which results in slow and memory intensive execution of these algorithms. Because of this, it is 
recommended that SSLClient have 8kb of memory available on the stack during a connection, and 4-10 seconds 
should be allowed for the connection to complete. Note that this requirement is based on the SAMD21—more 
powerful processors (such as the ESP32) will see faster connection times.
```

khoih-prog, 2022-04-06:
Close due to inaction.

closed 2022-04-06 by khoih-prog