khoih-prog/AsyncHTTPSRequest_Generic #12: setReuse feature

rjjrbatarao, 2022-10-18:
This library is great, one problem though when connecting twice on the same host takes a long time, I was wondering if theres a feature you could add to reuse the connection to avoid doing ssl handshake again, this could reduce the request time significantly  on second request.

khoih-prog, 2022-10-18:
Hi @rjjrbatarao 

Thanks for your nice words and the **enhancement** request.

I already found out and fixed the issue, and will release a new version within today. It won't reconnect if already connected, certainly after verifying the same host and port.

Even though your issue is a good one, **I almost ignored it** because the issue was opened without [**MRE**](https://stackoverflow.com/help/minimal-reproducible-example). Luckily, I changed my mind before pressing the button.

Hopefully you'll do it correctly next time.

Best Regards,

rjjrbatarao, 2022-10-24:
Wow thank you very much for the feature your the Man!, I'll definitely will follow the MRE next time, I'll test it with my current code and report how it goes.