khoih-prog/AsyncHTTPSRequest_Generic #20: Doesn't work on M5Stack devices (M5Atom & M5StickC Plus)

redphx, 2023-01-25:
### Describe the bug
This library doesn't work on M5Stack devices (M5Atom & M5StickC Plus). Constantly crashing, throwing `xQueueTakeMutexRecursive` errors.  

### Steps to Reproduce
```cpp
#define _ASYNC_HTTPS_LOGLEVEL_ 4

#include <WiFi.h>
#include <M5Atom.h>
#include <AsyncHTTPSRequest_Generic.h>

// Wifi Credentials
const char* WIFI_SSID = "wifi";
const char* WIFI_PASSWORD = "password";

void setup() {
  M5.begin(true, true, true);

  // WiFi Init
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("[WIFI] connecting to network " + String(WIFI_SSID) );
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("[WIFI] connected with Ip: " + WiFi.localIP().toString() );

  // divoom_client.LogIn();
  AsyncHTTPSRequest request;
  request.setDebug(true);
  static bool requestOpenResult;
  requestOpenResult = request.open("GET", "https://www.google.com/");
  request.send();
}

void loop() {
  M5.update();
}
```

### Expected behavior

A clear and concise description of what you expected to happen.

### Actual behavior
The script crashed the M5Atom constantly. It kept throwing these two errors:  

```
assert failed: xQueueSemaphoreTake queue.c:1549 (pxQueue->uxItemSize == 0)

Backtrace: 0x40083dd5:0x3ffd6800 0x4008be49:0x3ffd6820 0x4009171d:0x3ffd6840 0x4008ce59:0x3ffd6970 0x4008d010:0x3ffd69b0 0x400d41b4:0x3ffd69d0 0x400d4221:0x3ffd6a00 0x400d3e0e:0x3ffd6a20 0x400d48e1:0x3ffd6a50 0x400d48f1:0x3ffd6a80 0x400d4b7b:0x3ffd6aa0
```

```
0x40083dd5: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c line 402
0x4008be49: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c line 128
0x4009171d: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c line 85
0x4008ce59: xQueueSemaphoreTake at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c line 1545
0x4008d010: xQueueTakeMutexRecursive at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c line 731
0x400d41b4: AsyncHTTPSRequest::_onPoll(AsyncSSLClient*) at /Users/redphx/Documents/Arduino/libraries/AsyncHTTPSRequest_Generic/src/AsyncHTTPSRequest_Impl_Generic.h line 1622
0x400d4221: std::_Function_handler   >::_M_invoke(const std::_Any_data &, void *&&, AsyncSSLClient *&&) at /Users/redphx/Documents/Arduino/libraries/AsyncHTTPSRequest_Generic/src/AsyncHTTPSRequest_Impl_Generic.h line 1321
0x400d3e0e: std::function ::operator()(void*, AsyncSSLClient*) const at /Users/redphx/Library/Arduino15/packages/m5stack/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h line 260
0x400d48e1: AsyncSSLClient::_poll(tcp_pcb*) at /Users/redphx/Documents/Arduino/libraries/AsyncTCP_SSL/src/AsyncTCP_SSL_Impl.h line 1784
0x400d48f1: AsyncSSLClient::_s_poll(void*, tcp_pcb*) at /Users/redphx/Documents/Arduino/libraries/AsyncTCP_SSL/src/AsyncTCP_SSL_Impl.h line 2185
0x400d4b7b: _async_service_task(void*) at /Users/redphx/Documents/Arduino/libraries/AsyncTCP_SSL/src/AsyncTCP_SSL_Impl.h line 295
```
---
```

[AHTTPS] open( GET , url = https://www.google.com/
[AHTTPS] open: connecting to hostname = www.google.com:443
[AHTTPS] _client->connecting to www.google.com , 443
[AHTTPS] client.connect OK to www.google.com , 443
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008cff1  PS      : 0x00060e30  A0      : 0x800d41b7  A1      : 0x3ffd6990  
A2      : 0xa2413967  A3      : 0xffffffff  A4      : 0x00001004  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3ffb6c68  A8      : 0x80084195  A9      : 0x3ffd6980  
A10     : 0x3ffd15a8  A11     : 0x0000000c  A12     : 0x3ffc8b18  A13     : 0x3ffb26e0  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xa241396f  LBEG    : 0x40084bed  LEND    : 0x40084bf5  LCOUNT  : 0x00000027  

Backtrace: 0x4008cfee:0x3ffd6990 0x400d41b4:0x3ffd69b0 0x400d4221:0x3ffd69e0 0x400d3e0e:0x3ffd6a00 0x400d48e1:0x3ffd6a30 0x400d48f1:0x3ffd6a60 0x400d4b7b:0x3ffd6a80
```
```

PC: 0x4008cff1: xQueueTakeMutexRecursive at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c line 724
EXCVADDR: 0xa241396f

Decoding stack results
0x4008cfee: xQueueTakeMutexRecursive at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c line 717
0x400d41b4: AsyncHTTPSRequest::_onPoll(AsyncSSLClient*) at /Users/redphx/Documents/Arduino/libraries/AsyncHTTPSRequest_Generic/src/AsyncHTTPSRequest_Impl_Generic.h line 1622
0x400d4221: std::_Function_handler   >::_M_invoke(const std::_Any_data &, void *&&, AsyncSSLClient *&&) at /Users/redphx/Documents/Arduino/libraries/AsyncHTTPSRequest_Generic/src/AsyncHTTPSRequest_Impl_Generic.h line 1321
0x400d3e0e: std::function ::operator()(void*, AsyncSSLClient*) const at /Users/redphx/Library/Arduino15/packages/m5stack/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h line 260
0x400d48e1: AsyncSSLClient::_poll(tcp_pcb*) at /Users/redphx/Documents/Arduino/libraries/AsyncTCP_SSL/src/AsyncTCP_SSL_Impl.h line 1784
0x400d48f1: AsyncSSLClient::_s_poll(void*, tcp_pcb*) at /Users/redphx/Documents/Arduino/libraries/AsyncTCP_SSL/src/AsyncTCP_SSL_Impl.h line 2185
0x400d4b7b: _async_service_task(void*) at /Users/redphx/Documents/Arduino/libraries/AsyncTCP_SSL/src/AsyncTCP_SSL_Impl.h line 295
```

### Information
```
Arduino IDE version: 1.8.19
M5Stack Core Version 2.0.6
OS: macOS Catalina 10.15.7
```

khoih-prog, 2023-01-25:
HI @redphx 

Thanks for your interest in the library.

Sorry that I have no M5Stack device to test to know what wrong with the design / implementation (hardware, firmware , software, etc.). Start with a standard ESP32 to see if your code is OK. Transfer to M5Stack device only after everything is OK with ESP32.

You have to test, experiment to figure the culprit out. If the issue is caused by bug of this library, please inform or make PRs to fix.

Good Luck,

closed 2023-01-25 by khoih-prog

redphx, 2023-01-26:
Hi @khoih-prog,
The code above also didn't work on Lilygo T-Display-S3 (ESP32-S3). But I think I found the problem.

It works when defining `request` as global variable:  
```cpp
AsyncHTTPSRequest request;
void setup() {
    ...
}
```
But it won't work this way (constantly crashing):  
```cpp
void setup() {
    ...
    AsyncHTTPSRequest request;
    ...
}
```

Here is the full sketch for Lilygo T-Display-S3.

```cpp
#define _ASYNC_HTTPS_LOGLEVEL_ 4

#include <WiFi.h>
#include "Arduino.h"
#include <AsyncHTTPSRequest_Generic.h>

// Wifi Credentials
const char* WIFI_SSID = "wifi";
const char* WIFI_PASSWORD = "password";

// It works when I define "request" here
// AsyncHTTPSRequest request;

void requestCB(void *optParm, AsyncHTTPSRequest *request, int readyState)
{
  (void) optParm;

  if (readyState == readyStateDone)
  {
    AHTTPS_LOGDEBUG0(F("\n**************************************\n"));
    AHTTPS_LOGDEBUG1(F("Response Code = "), request->responseHTTPString());

    if (request->responseHTTPcode() == 200)
    {
      Serial.println(F("\n**************************************"));
      Serial.println(request->responseText());
      Serial.println(F("**************************************"));
    }

    request->setDebug(false);
  }
}

void setup() {
  Serial.begin(115200);

  // WiFi Init
  WiFi.mode(WIFI_STA);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("[WIFI] connecting to network " + String(WIFI_SSID) );
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("[WIFI] connected with Ip: " + WiFi.localIP().toString() );

  // It crashes when I define "request" here
  AsyncHTTPSRequest request;

  request.setDebug(true);
  request.onReadyStateChange(requestCB);

  static bool requestOpenResult;
  requestOpenResult = request.open("GET", "https://worldtimeapi.org/api/timezone/America/Toronto.txt");
  request.send();
}

void loop() {
}
```

Is it a bug or something I didn't know of?  
Thank you for the library.

khoih-prog, 2023-01-26:
> // It crashes when I define "request" here
>   AsyncHTTPSRequest request;

It'll certainly crash because when `setup()` ends, the `request` object will be destroyed along with all the necessary info / data / pointers, etc.

redphx, 2023-01-26:
That makes sense. Thank you for your help.