khoih-prog/AsyncHTTPSRequest_Generic #21: Bug with _parseURL()

redphx, 2023-01-25:
`_parseURL()` doesn't consider `https://www.google.com` a valid URL, but it does with `https://www.google.com/`.

khoih-prog, 2023-01-25:
You have to save us some time by including [**MRE**](https://stackoverflow.com/help/minimal-reproducible-example) in any issue. I certainly can create some code to test, but just wasting time, better to use yours.

redphx, 2023-01-26:
Sorry, here is the sketch:  

```cpp
#define _ASYNC_HTTPS_LOGLEVEL_ 4

#include <WiFi.h>
#include "Arduino.h"
#include <AsyncHTTPSRequest_Generic.h>

// Wifi Credentials
const char* WIFI_SSID = "wifi";
const char* WIFI_PASSWORD = "password";

AsyncHTTPSRequest request;

void setup() {
  Serial.begin(115200);

  // WiFi Init
  WiFi.mode(WIFI_STA);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("[WIFI] connecting to network " + String(WIFI_SSID) );
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("[WIFI] connected with Ip: " + WiFi.localIP().toString() );

  request.setDebug(true);

  request.open("GET", "https://www.google.com");
  // request.open("GET", "https://www.google.com/");
}

void loop() {
}

```
  
When URL = `https://www.google.com`
```
[AHTTPS] open( GET , url = https://www.google.com
[AHTTPS] open: error parsing URL
```  

When URL = `https://www.google.com/`  
```
[AHTTPS] open( GET , url = https://www.google.com/
[AHTTPS] open: connecting to hostname = www.google.com:443
[AHTTPS] _client->connecting to www.google.com , 443
[AHTTPS] client.connect OK to www.google.com , 443
[AHTTPS] _onError handler SSL error = OK
```

khoih-prog, 2023-02-01:
Hi @redphx  

The new [**AsyncHTTPSRequest_Generic** v2.5.0](https://github.com/khoih-prog/AsyncHTTPSRequest_Generic/releases/tag/v2.5.0) has just been published. Your contribution is noted in [**Contributions and Thanks**](https://github.com/khoih-prog/AsyncHTTPSRequest_Generic#contributions-and-thanks)

Please test and report more bugs.

Best Regards,

---

### Releases v2.5.0

1. Add support to ESP32 boards using `LwIP W6100 Ethernet`
2. Fix bug of wrong `reqStates`. Check [Callback behavior buggy #19](https://github.com/khoih-prog/AsyncHTTPSRequest_Generic/issues/19)
3. Fix bug of `_parseURL()`. Check [Bug with _parseURL() #21](https://github.com/khoih-prog/AsyncHTTPSRequest_Generic/issues/21)
4. Improve `README.md` so that links can be used in other sites, such as `PIO`

khoih-prog, 2023-02-01:
These are the terminal outputs using your test code

#### Using url = https://www.google.com

```
[WIFI] connecting to network HueNet.......
[WIFI] connected with Ip: 192.168.2.151
[AHTTPS] open( GET , url = https://www.google.com
[AHTTPS] pathBeg= -1
[AHTTPS] hostEnd= 22
[AHTTPS] portBeg= -1
[AHTTPS] _URL->host= www.google.com
[AHTTPS] _URL->path= 
[AHTTPS] _URL->query= 
[AHTTPS] open: connecting to hostname = www.google.com:443
[AHTTPS] _client->connecting to www.google.com , 443
[AHTTPS] client.connect OK to www.google.com , 443
[AHTTPS] _onError handler SSL error = OK
```

#### Using url = https://www.google.com/

```
[WIFI] connecting to network HueNet.......
[WIFI] connected with Ip: 192.168.2.151
[AHTTPS] open( GET , url = https://www.google.com/
[AHTTPS] pathBeg= 22
[AHTTPS] portBeg= -1
[AHTTPS] _URL->host= www.google.com
[AHTTPS] _URL->path= /
[AHTTPS] _URL->query= 
[AHTTPS] open: connecting to hostname = www.google.com:443
[AHTTPS] _client->connecting to www.google.com , 443
[AHTTPS] client.connect OK to www.google.com , 443
[AHTTPS] _onError handler SSL error = OK
```

closed 2023-02-01 by khoih-prog